Akaflieg-Freiburg/enroute #23: Make the app compatible to pickup traffic information

shortyxs, 2020-02-19:
Great work with the App so far!

I'd  like to see traffic beeing displayed from a source, such as Stratux.
The team ofthe [Stratux Europe Edition](https://github.com/b3nn0/stratux), is doing an awsome job, gaining the information via ADS-B and  Flarm, to provide it to a Map App, like enroute.
As Straux is also Open Source, I'd prefere this over any other commercial solution.

kebekus, 2020-02-20:
Dear Mr. Schroeder,

thank you for this suggestion. I have not yet made up my mind if (and to what extend) I will include this feature, as there are many outstanding issues that are probably more urgent. I shall come back to you in a while. Would you be willing to help with the implementation?

Best - Stefan Kebekus.

joachimwe, 2021-03-13:
The Stratux Support is included since 2.5

kebekus, 2021-03-15:
As joachimwe pointed out: Stratux devices are supported now. I am closing this issue. Thank you for your help!

closed 2021-03-15 by kebekus