Akaflieg-Freiburg/enroute #28: altitude seems to be systematically wrong

think-nice-things, 2020-03-14:
Hi,

the altitude seems to be systematically wrong, compare for example with the known altitude of your current location or other apps like gps status. For my home location the offset is about 45 m which corresponds to the geoid correction.
Do we need to apply a geoid correction?

kebekus, 2020-03-15:
Thank you for your report. I never thought of geoid corrections and always considered GPS height data to be unreliable. For now, you can correct your altitude manually (have a look at the menu), but I will be happy to implement geoid correction. Could you please give me a reliable reference where I can find the formulas to implement? -- Best, Stefan.

think-nice-things, 2020-03-15:
Hi Stefan,

as far as I can see there are different geoid modes available either as spherical harmonics or gridded data.

Two good starting points are
* https://geographiclib.sourceforge.io/1.18/geoid.html
* https://earth-info.nga.mil/GandG/update/index.php?action=home

at https://geographiclib.sourceforge.io/1.18/geoid.html, they apparently generated gridded data from https://earth-info.nga.mil/GandG/update/index.php?action=home. It's also explained there in detail to which accuracy the different models are good for.

A more lightweight example is at [geoid.py](https://gist.github.com/pyRobShrk/98c43c25d4d3c5d105e90fa0b58cf823) which uses the [EGM96](https://earth-info.nga.mil/GandG/wgs84/gravitymod/egm96/binary/binarygeoid.html) gridded data and interpolates with splines.

In the app [Locus Map](https://play.google.com/store/apps/details?id=menion.android.locus) you can explicitely turn on or off the geoid correction (see configuration "height manager"). Apparently the correction is also done with the [EGM96](https://earth-info.nga.mil/GandG/wgs84/gravitymod/egm96/binary/binarygeoid.html) data as Locus installs [WW15MGH.DAC](https://earth-info.nga.mil/GandG/wgs84/gravitymod/egm96/binary/WW15MGH.DAC) to /storage/emulated/0/Locus/data/srtm.

think-nice-things, 2020-03-15:
there might be a simpler solution using GGA fix data from NMEA sentences, see https://www.gpsinformation.org/dale/nmea.htm

kebekus, 2020-03-16:
Ok, I will look into this. The issue has however low priority because the 'altitude correction dialog' offers a clean way for the user to work around the altitude correction problem, in a way that pilots are used to.

kebekus, 2020-05-27:
Feature is included in version 1.7.0, which will be distributed shortly.

closed 2020-05-27 by kebekus