Akaflieg-Freiburg/enroute #41: No support for diacritic used in some European countries

siren15, 2020-05-24:
No support for fonts used in some European countries that include support for letters that use caron( ̌ ).
For example ľ č ť ž and more. This issue breaks a names for a lot of cities in balto-slavic countries.

kebekus, 2020-05-24:
Dear Siren15, I do not fully understand where the problem is. Is it in the base map display? Or perhaps in the names of airspaces or waypoints? Could you please give me more complete information, and perhaps also include a screenshot? I am not from a balto-slavic country, so I have no idea about the proper spelling of names there…

siren15, 2020-05-24:
I'm sorry if I worded it poorly before. The names of some cities in the map don't have diacritic at all. Mainly when it comes to letters that use caron( ̌ ). The letters are missing so the names are broken. I'm posting some screenshots from your app and then some google maps screenshots for reference on how it should be. I hope this will help. Thanks.
[image]
[image]
[image]
[image]