Akaflieg-Freiburg/enroute #48: EditText is not displaying while modyfing flight route

SKocur, 2020-06-25:
Hello, I have an issue with Enroute app installed on Samsung Galaxy A8 (EditText is not displaying when user wants to edit route data). It works on Samsung Galaxy S7.

Steps to reproduce (on Samsung Galaxy A8):
* clean install app
* open navigation drawer menu
* click "Route"
* swipe to "Aircraft and ..." tab
* try to edit anything

It does not work even if you specify waypoints.

kebekus, 2020-06-26:
Hello, this is bad. I do not have access to a Galaxy A8 device and have not been able to reproduce the issue. Would you be able to produce a little screencast video of the failure, so I can see how exactly the issue looks like?

SKocur, 2020-06-26:
Yes, sure, here you are

SKocur, 2020-06-26:
[image]

kebekus, 2020-06-26:
I understand! The display is so narrow that the text fields have size zero! Can you quickly check if it help to turn the device sideways, so that there is more horizontal screen space?

SKocur, 2020-06-26:
Sorry for late respose. You are right, this field is displayed correctly if you use app in horizontal orientation.

kebekus, 2020-06-26:
Very well. I will try to have a bugfix in place for the next release. Thank you for reporting this.

SKocur, 2020-06-26:
Thank you for doing great job! Keep it up :)

kebekus, 2020-07-07:
Dear SKocur,

a little unrelated to your bug report … you seem very competent with computers. Are you interested in maintaining a Polish translation of our app? Also, we are constantly looking out for competent contributors…

Best,

Stefan

kebekus, 2020-07-07:
Bug has been fixed. Bugfix will be included in version 1.8.0, scheduled to appear in a month or so.

closed 2020-07-07 by kebekus

SKocur, 2020-07-07:
Hello Stefan, I would like to become contributor. We can talk via e-mail: skocur10{at}gmail[dot]com

Glad that you fixed it! :)