Akaflieg-Freiburg/enroute #52: Move route entries up and down with drag'n'drop

HeinerTholen, 2020-07-06:
Title says it all. 

If you want to add a waypoint into the middle of a long route, it's quite cumbersome at the moment, using the three-dot-menu.

kebekus, 2020-07-07:
I totally agree. Would you be interested in implementing this feature?

HeinerTholen, 2020-07-08:
Yes, happy to do this. But I'd leave it open for now, as I'm planning to look into the flight log next.

kebekus, 2022-03-19:
TTC70 wrote: In a flight plan with, e.g. 10 waypoints, if you want to add a waypoint in second position, you have to manually move it up 8 positions in the flight plan.

Heiner suggested an improvement that would solve the issue. A solution might be [here](https://doc.qt.io/qt-5/qtquick-tutorials-dynamicview-dynamicview1-example.html) but I have not had the time to implement this.

@TTC70 Let's use this communication thread for flight plan reordering.

kebekus, 2026-09-15:
Version 3.5.0 is now available on Google Play, and the iOS version will follow shortly. Waypoints in the flight route can now be reordered by drag-and-drop. (I had announced this feature for 4.0.0 above; that release became 3.5.0.)

@HeinerTholen @TTC70 Thank you for the suggestion and for your patience! If anything does not work as it should, please keep the reports coming.