Akaflieg-Freiburg/enroute #54: Weather feature

acrovato, 2020-07-09:
Hello everyone,

I plan to start implementing a new feature allowing to get METAR and TAF weather data.
Here is a description of the proposed implementation. Note that I am not familiar with the code, so the implementation I describe here might change as I code. I look forward to reading your comments or suggestions, should you have any.

**QML**
An entry "Weather" will be added to the drawer. On click, it will open a new `Weather` page with:
- a scrolling list showing the different aerodromes
- a search bar allowing to filter the aerodromes by name or ICAO code

All in all, this page should look similar to the `FlightRouteAddWPDialog` dialog.
When an aerodrome is selected, a new `Weather` dialog will be opened. This dialog will consist of three sections:
- Epheremis: two rows showing the sunrise and sunset time
- METAR: one row showing the raw METAR data
- TAF: one row showing the raw TAF data

The dialog will have two buttons: `Refresh` and `Close`.  
Additionally, a colored badge displaying the flight rules (VFR, MVFR, ...) and a row showing the time at which the data was fetched will be added.

**C++**
The logic powering the `Weather` page and dialog described above will be held in the class `Meteorologist`.
The class will be constructed using a `Waypoint` (an aerodrome). It will have the following members:
- `file`, the file where the weather data will be saved to disk (Appdata/weather/icao.json, where icao is the ICAO code of the aerodrome)
- `sunrise`, the sunrise time
- `sunset`, the sunset time
- `metar`, the raw metar data
- `taf`, the raw taf data
- `rules`, a pair containing the flight rule and its associated color
- `fTime`, the time at which the weather data were fetched

Each of these members will be private and accessible through dedicated getters.
Additionally, the class will have the following methods:
- `read()`, read the weather data from disk and set the members accordingly
- `computeTimes()` compute the sunrise and sunset times and set the members accordingly
- `fetch()`, fetch the weather data from NOAA, set the members accordingly and save to disk

A `Meteorologist` object will be constructed when an aerodrome is clicked on the `Weather` page, and destroyed when the `Close` button is clicked.
The constructor will call `read()` and `computeTimes()`. A click on the `Refresh` button will call `fetch()`.

kebekus, 2020-07-11:
Dear Adrien,

I agree in principle with your plan for implementation.  There are, however, a
few things that I would suggest to do differently.

### GUI Design

I am fine with the weather page. But what airfields should be shown? I would
suggest the following design.

- The app requests and shows METAR/TAF reports for all stations within 75NM from
  the current position and the current flight route. I presume you want to use
  https://www.aviationweather.gov/dataserver ?

- At the top of the page, there is a slider for "automatic updates". When
  switched to on, the app tries to update weather report at regular intervals
  (say: all 10 min when in flight and all 30 min when on the ground). When
  automatic updates are enabled, there should be a very visible Android
  Notification.

- Additionally, there should be a mechanism to manually update the weather info.
  This could be very visible as long as the list is empty, and otherwise follow
  the example of the page "Settings/Library/Map".

- I am obsessed with privacy. Before any data is fetched, I would therefore like
  to have a dialog that warns the user that using this feature will forward
  information about the user's position and flight route to a US government web
  site.  The dialog should of course have a button "Do not show this dialog
  again".

- Most people cannot read METAR/TAF raw data. There should be some logic to
  translate the raw coded to natural languages!

- I am not convinced that we need sunrise/sunset for all airfields. Given the
  realistic speed of our aircraft, it is enough for all practical purposes to
  have sunrise/sunset for the current position.  The natural place for this
  would be the top of the "weather" page.

### C++ design

I think that we need more classes.

- The class Meteorologist contains the logic for fetching/updating METAR/TAF
  reports.  This class also exposes lists of reports to the QML-part of the app.

- A new class "Report" that contains the individual reports. These classes are
  owned, generated and deleted by the Meteorologist.  The "Report" class could
  also be the place for the logic to translate the raw coded to natural
  languages.

- I do not see the need to store any files on the disk. The "Report" classes can
  contain the required data.

Best,

Stefan.

kebekus, 2020-08-12:
Implemented, will appear with the next version.

closed 2020-08-12 by kebekus