Akaflieg-Freiburg/enroute #62: Add waypoint library

kebekus, 2020-07-23:
The title says it all.
Markus Sachs also suggest: "Wegpunkte allgemein: Das Angebot "auf Karte anzeigen" wäre eine schöne Ergänzung. Ein per Antippen zur Strecke hinzugefügter Wegpunkt lässt sich nicht sinnvoll benennen und nicht als Favorit abspeichern. "

HeinerTholen, 2020-07-24:
I don't quite understand this. Should it be a list of waypoints that the user had in a route in the past? Are these favorite waypoints?

kebekus, 2020-07-24:
This is meant to be a library of waypoints that the users choses to save for use in future flight routes. Typical use cases:

- Landmarks (such as the Hohenzollern Castle) that the pilot might wish to fly to for pleasure, or that help with visual navigation.

- Landing strips that do not appear in the map.

HeinerTholen, 2020-07-27:
Ah, I see. Yeah, this is nice.

woodwind-enr, 2020-07-27:
Hi Heiner,
don't know whether my email response reached you; github did not assign it to #62. Here what I wrote:

(1) "Should it be a list of waypoints that the user had in a route in the past?"

--> that was not my suggestion but would nevertheless be a nice feature: a list of recently used waypoints, with the most recently used ones on top of the list. However, only makes sense if waypoints can be given names different from "waypoint" (see below).

(2) "Are these favorite waypoints?" --> Not automatically. They could become favourites as soon as a favourite function is implemented. So the "Favourites" list would be different from the "Recent Waypoints" list.

My suggestions in this context are:

(a) "Show on map" function for each waypoint, no matter whether as part of a route or stored as favourite.

(b) "Rename" function for self-made waypoints (created by finger tip on the screen or by input, see below).

(c) "Favourite" functions, which means the usual show, store, edit/rename and delete mechanisms.

Advanced features (whenever there might be time to implement them):

(d) Create waypoint by entering latitude and longitude, e.g. after copying from Google Maps, OSM etc. or from a friend who sends the coordinates of his landing meadow. Maybe not that important for landing purposes but for creating own waypoints around big CTRs, forests or lakes. (Sorry, I come from ultralight where most engines are not certified and prudent pilots might wish to avoid unlandable terrain while planning their routes.)

(e) Import waypoints from external list (text file in predefined format).

BTW, off-topic: there should be a general decision (Stefan!) for British or American English, e.g. favourites vs. favorites.

closed 2022-06-21 by kebekus