Akaflieg-Freiburg/enroute #68: User provided waypoints

obetz, 2020-08-02:
User provided waypoints e.g. in the CUP file format used by Naviter should be supported

kebekus, 2020-08-03:
Thank you for your comment. This feature is already on our TODO-list. But again, we are a small group of people who develop the app in our spare time, so please expect that this will take a long time (unless you are willing to help with the implementation).

kebekus, 2022-06-21:
Dear all,

Thank you for your suggestions, even if they took me a long time to implement. Version 2.18.0 of **Enroute Flight Navigation** will feature a "waypoint library." Users can add waypoints via the moving map or import a file in CUP, GeoJSON, or GPX format. 

Thanks again. I hope you are still enjoying the app.

Stefan.

closed 2022-06-21 by kebekus