Akaflieg-Freiburg/enroute #69: User provided airspace

obetz, 2020-08-02:
User provided airspace data e.g. in "OpenAir" format should be possible, e.g. for local noise prevention zones or for airspaces from NOTAM. See http://www.winpilot.com/UsersGuide/UserAirspace.asp for (kind of) a specification of the OpenAir format.

kebekus, 2020-08-03:
Thank you for your comment. It is quite likely that the feature that you request will be implemented. However, we are a small group of people who develop the app in our spare time, so please expect that this will take a long time.

WilliPrestl, 2020-08-17:
We also would appreciate userprovided airspaces for local noise prevention very much. OpenAir Format would be ok for us. Example attached.
[3.14 KDF MM 2020-07-02.txt](https://github.com/Akaflieg-Freiburg/enroute/files/5084806/3.14.KDF.MM.2020-07-02.txt)

kebekus, 2020-11-15:
Update: Adrien Crovate has kindly volunteered to work on this. The implementation will take it's time, tough.

kebekus, 2021-04-09:
Adrian gave up on the job due to lack of time.

acrovato, 2021-04-09:
Yes, I do not have much time now, unfortunately...  
For the record, here are two repo that contains logic which could be used as a starting point:
https://github.com/Turbo87/aerofiles/tree/master/aerofiles
https://github.com/dbrgn/openair-rs
Note that the first repo also contains additional resources for .cup data file (not well documented).

kebekus, 2023-04-17:
Dear all: Heinz Blöchinger has volunteered to work on this project. This is extremely good news.

@bloech1 You might want to look at the comments of @acrovato, who points to some helpful resources…

kebekus, 2023-06-22:
Thanks to the work of Heinz we're making progress:

[image]

kebekus, 2023-06-28:
@WilliPrestl Danke für die Rückmeldung. Ich würde den Fall "Lärmschutzgebiete für Königsdorf" gern für Förderanträge/Pressemitteilungen/etc verwenden. Ist das OK für Euch und würdet Ihr das unterstützen?

Falls ja: sind die Lärmschutzzonen, die in der letzten Karte angezeigt wurden, korrekt oder gab es inzwischen Änderungen? Es wäre super-ärgerlich, wenn ich eine Pressenotiz mache und am Ende kommt raus, dass die eingezeichneten Zonen gar nicht stimmen…

obetz, 2023-06-28:
ich bin stark dagegen, die Lärmschutzzonen von Königsdorf öffentlich zu machen oder auch nur öffentlich darauf hinzuweisen.

kebekus, 2023-07-03:
@obetz Die Nachricht ist angekommen, auch wenn mich dein Ton irritiert. Wir haben uns sehr viel Mühe gegeben, das Feature für euch in unserer Freizeit zu implementieren. Da hätte ich einen etwas netteren Ton erwartet.

kebekus, 2023-07-03:
@obetz @WilliPrestl Das Feature ist implementiert und wird in der kommenden Programmversion erscheinen, vermutlich gegen Ende der Woche. Ich schließe diese Diskussion daher.

closed 2023-07-03 by kebekus

obetz, 2023-07-03:
> @obetz Die Nachricht ist angekommen, auch wenn mich dein Ton irritiert. Wir haben uns sehr viel Mühe gegeben, das Feature für euch in unserer Freizeit zu implementieren. Da hätte ich einen etwas netteren Ton erwartet.

ich verstehe nicht, wie Du meine Aussage persönlich nehmen oder auch nur als unfreundlich empfinden kannst. Es geht nur um die Veröffentlichung der Daten, nicht Eure Leistung.

Nehmt doch für Pressemitteilungen einfach NOTAM-Lufträume, so wie ich es 2020-08-02 im OP vorgeschlagen habe. Das ist für alle Piloten wichtig (ich mache vor jedem Flug aus den NOTAM-Lufträumen ein OpenAir-File und verwende es in LK8000).