Akaflieg-Freiburg/enroute #81: ETA/Distance to next WP

spuckebaer, 2020-09-01:
Not sure if not allready mentioned.

Just a simple measurement of distance towards next waypoint and estimated time of arrival would be useful

Dzordzik, 2021-03-10:
Yes, I miss this features too. Please implement it. It may be one of basic features for navi app.

Dzordzik, 2021-03-10:
I´m missing ETE next, ETE destination and ETA for both, heading and distance to next WP and distance to destination. Ideally with menu settings what I want to see and what not to best screen space usage.

kebekus, 2022-03-05:
Dear spuckebaer and Dzordzik,

thank you for your suggestion! I am starting the implementation now. Hopefully, I will have a beta version within a few weeks.

Best,

Stefan.

spuckebaer, 2022-03-06:
Thanks very much. If it helps, I'd be happy to test the beta

kebekus, 2022-03-15:
Dear all,

The next release of **Enroute Flight Navigation** is coming up, and much has changed internally. I expect a longer-than-usual beta test phase. I would greatly appreciate your feedback if you could try the new features. I am particularly interested to learn if the app reliably recognizes the next waypoint when you follow a route (critical cases might include: routes with sharp angles, return routes, switching the app on and off during the flight, switching the app off before a waypoint, )

Best wishes,

Stefan.

-----

## New features 

### ETE, ETA, and rubber band 

If you fly along a route, the app will show the remaining distance, ETE, and ETA for the next waypoint and the final waypoint. The moving map shows a red "rubber bad" between the current position and the next waypoint. The information is shown under the following conditions:

- The user has set a route.
- The user has set aircraft cruise speed and wind information.
- The aircraft is flying.
- The plane is closer than three nautical miles to the route.
- The distance to the final waypoint is more than three nautical miles.

### Behavior on direct-to

When double-tap into the moving map, the waypoint description dialog will open, and you have the choice to set a quick "direct to" route. The app will now show the rubber band to the waypoint when you do that. The following conditions must be met:

- The user has set aircraft cruise speed and wind information.
- The aircraft is flying.
- The distance to the final waypoint is more than three nautical miles.

### Map update notifications

The app now uses Android notifications to alert you when map updates are available. The alert will not show up while in flight.

spuckebaer, 2022-03-25:
Hi Stefan,

 

Thanks for your initiative.I have tried on both devices (Samsung S9 and Lenovo Tab) with 2.15.93 and could in both case git no indications.

 

Flight 1 EDHE - EDWG

Flight 2 EDHE - EDXM

 

Any hint for me or info you need?

 

KR

 

Jens

 
 

Gesendet: Dienstag, 15. März 2022 um 07:27 Uhr

kebekus, 2022-03-28:
Dear all,

Thank you for your feedback! I understand from all the reports that many users are confused because the Remaining Route Infobox switches on and off during the flight without apparent reason. Some users also requested that the Infobox is visible on the ground, before take-off.

I have improved the app accordingly and uploaded a new version (=2.15.94) to Google Play. Instead of switching the Infobox off, the app now shows a reason why the box is no longer visible. Also, the box is shown before take-off (except start point = endpoint, because then the app cannot tell if we are "before start" or "after landing"). If you have the chance, please test and send me your feedback!

Best wishes · Thank you for the help,

Stefan.

kebekus, 2022-04-06:
I am closing this issue because the request is implemented in 2.16.0. Thank you for your help!