Akaflieg-Freiburg/enroute #85: Automatically zoom in on flying in the circuit (or near the airfield)

lucabert, 2020-09-08:
It should be very helpful if the App automatically zoom in on nearing the airfield (ideally flying in the circuit).
Since this is a very tricky moment in the fly, the pilot does not have many time to zoom in to see the circuit.

kebekus, 2020-11-07:
Dear Luca,

I experimented a bit and found that most users get very confused when the zoom changes automatically. I would therefore rather not implement this feature, and ask for your understanding.

Thanks!

Stefan.

closed 2020-11-07 by kebekus

lucabert, 2020-11-07:
Am 07.11.2020 um 11:02 schrieb Stefan Kebekus:

Hi Stefan,

> I experimented a bit and found that most users get very confused when
> the zoom changes automatically. I would therefore rather not implement
> this feature, and ask for your understanding.

That's not a problem!

It was just a suggestion.

Regards
Luca