Akaflieg-Freiburg/enroute #86: Integration of a Flarm device?

markus2marks, 2020-09-09:
This project is very nice!! Do you also want to integrate an interface for a flarm device?

kebekus, 2020-09-10:
I do have plans for integrating Flarm. This is, however, a long-term project that will take some time … so please be patient with us

spuckebaer, 2020-12-11:
Would be great to include transponder signals aswell :-) -maybe interface to Stratux?

kebekus, 2021-02-19:
Hi, 

a few minutes ago I pushed a technology preview for our app to Google Play. The app can now connect to your aircraft's traffic receiver (typically a FLARM device) and show nearby traffic on the moving map. 

@spuckebaer Transponder signals are supported. Stratux might just work, but I do not have a Stratux device to test it out. 

Open the menu and go to Information/Traffic Reveicer to try.

Please send me your feedback! Reach me by e-mail, GitHub, or via any other channel.

- Does the feature work for you at all? Do you run into problems?
- If it works, how do you like the map design?
- What traffic receiver do you use?
- Was the setup difficult? Did you find the help texts helpful? Did you even find the help texts?

Best,

Stefan.

closed 2021-02-19 by kebekus

spuckebaer, 2021-02-19:
Hi Stefan,

 

perfect, thx :-)

 

2 days ago I was flying with club mates and convinced them to use enroute also. There only complain was: No flarm/stratux/ADS-B 

 

Even if I live quite close to HH airport: Currently no traffic, so I could just check that Stratux and Enroute can get connected -> it works.

 

Only issue I had: A switch from 192.168.10.1 to 192.168.1.1 within Stratus was neccassary.

 

I guess it might be usefull to enable the end user to adapt "IP Source" within the settings. For 90% of Stratux users it is no problem to adapt, but as it is more likely to move across different planes with the tablet than the receiving device.

 

..I will try out this weekend and reply to you how it works with real traffic. Will bearingles targets also be displayed?

 

New map design is really good, I like the contrast. Only the nature reserves (in green) is too dominant.

 

Helptext was fine for me...was not expecting to fins them in infirmation (...settings would be more logical to me, but o.k.)

 

 

Again: THX very good too and help I am using in 95% of my flights today.

 

KR

 

Jens

 

 

 

 
 

Gesendet: Freitag, 19. Februar 2021 um 07:37 Uhr

spuckebaer, 2021-02-20:
Hi Stefan,

 

first update....seems to work :-) flarm antenna was not really adjusted on my side and really no further traffic . (Just some bearingless targets, would be nice to add those to the map aswell)

 

 

Had some crashes and connection lost to stratux, but could not really identify why. (Restart recovered all times)  Anything I can provide as info to ease your analysis?

 

 

Green arrow of the target might be not easy to see when flying on a sunny day.

 

But: Really perfect, very good way forward...thanks

 

KR

 

Jens

 

 

Hi Stefan,

 

perfect, thx :-)

 

2 days ago I was flying with club mates and convinced them to use enroute also. There only complain was: No flarm/stratux/ADS-B 

 

Even if I live quite close to HH airport: Currently no traffic, so I could just check that Stratux and Enroute can get connected -> it works.

 

Only issue I had: A switch from 192.168.10.1 to 192.168.1.1 within Stratus was neccassary.

 

I guess it might be usefull to enable the end user to adapt "IP Source" within the settings. For 90% of Stratux users it is no problem to adapt, but as it is more likely to move across different planes with the tablet than the receiving device.

 

..I will try out this weekend and reply to you how it works with real traffic. Will bearingles targets also be displayed?

 

New map design is really good, I like the contrast. Only the nature reserves (in green) is too dominant.

 

Helptext was fine for me...was not expecting to fins them in infirmation (...settings would be more logical to me, but o.k.)

 

 

Again: THX very good too and help I am using in 95% of my flights today.

 

KR

 

Jens

 

 

 

 
 

Gesendet: Freitag, 19. Februar 2021 um 07:37 Uhr