Akaflieg-Freiburg/enroute #87: Suggestion

jordirar, 2020-09-17:
In the flight path section> ACFT and wind, the descent speed does not work.  Say, when you generate a route from A to B, the only speed you take is Cruise. If I modify the value of the descent speed, the total values, time and liters are the same.  
If they correct that, the TOD and TOC points could be included for VFR navigation, when one goes from climb to cruise, and from cruise to descent.

The app in general is very good.  It is easy, simple and intuitive.  They could include some more functions, which are used for precomputer. Something to do with flight level.  Or that one suggests, taking into account the departure and arrival airport, they may have different elevación. Thanks and regards

jordirar, 2020-09-17:
Another suggestion may be that when you touch the North button (N), the map is oriented in that north direction.  Like GPS (Google maps or Here maps)

kebekus, 2020-09-17:
Dear Jordinar,

thank you for your suggestions. I am closing this request because it duplicates other requests, in particular #16. As you observed, the descent speed is not used at present, this will change in future versions of the app.

Development is slow, because the app is developed by a small group of flight enthusiasts in our spare time -- so please do not expect the features to come up quickly. If you are willing to help with the programming, let us know.

Best wishes,

Stefan.

closed 2020-09-17 by kebekus