Akaflieg-Freiburg/enroute #88: Please include flight sim gps source data

ricepmk, 2020-09-17:
As title says. Thanks!

kebekus, 2020-09-18:
Dear Ricepmk,

I understand your request. However, this app concentrates real flight, so adding a flight simulator GPS source is not high on our priority list. I am closing this issue because I cannot see that we have to resources to implement this feature anytime soon. I understand that this must come across as a disappointment, and I can only ask for your understanding.

Note however that this is an open source project -- anybody has access to the source code and can start coding today. If you or someone else is willing to implement this feature, then please contact me directly and I am happy to support the implementation.

Best wishes,

Stefan.

ricepmk, 2020-09-18:
Stefan, thanks for taking the time to respond and congratulations on such a wonderful open source app. I understand what you are saying, but hopefully someday Enroute will have this option if anyone in the community decides to collaborate.

Best wishes,

Ricardo Escudé
PM [a] alia dot pro

kebekus, 2021-05-05:
Dear Ricepmk,

the latest beta version of the app (=2.6.9) should support flight simulators now. It has been tested with X-Plane 11, but I do not have a Windows machine, so I cannot test MS Flight Simulator, which is however arguably the most important program around.

Can you please test the app and tell me if/how well things work for you? I wrote a draft manual, which you can find here

- https://akaflieg-freiburg.github.io/enrouteText/experimentalmanual/html/02-steps/simulator.html

The article 

- https://support.foreflight.com/hc/en-us/articles/204115275-How-do-I-connect-Microsoft-Flight-Simulator-FS-X-or-FS-2004-to-ForeFlight-

might be helpful in setting up the simulator to work with the app. Any help is greatly appreciated. Thanks again for your support!

Best

Stefan.

kebekus, 2021-05-12:
Closing because Ricepmk did not reply.

closed 2021-05-12 by kebekus