Akaflieg-Freiburg/enroute #89: Airport shapes

darkm20, 2020-09-18:
Is there any chance to have on the map the airport shapes ? I think they should be on the layer 11.
Thanks!

kebekus, 2020-09-21:
Dear darkm20,

thank you for your report. The feature that you request will be difficult to implement, and is not very high on our priority list. With the limited resources that I have, I cannot see that I will implement it soon. I will therefore close this issue now and ask for your understanding.

That said -- the app is 100% open source and everybody is invited to contribute. If you are interested to implement this feature, I will be glad to offer advice.

Best,

Stefan.

closed 2020-09-21 by kebekus

darkm20, 2020-09-21:
Hello Stefan, thanks for answering. Any advice from you would be appreciated.
Is it something related to the mbtiles ? Do I need to generate new mbtiles for each country with layer 11 (airport shapes)?

Thank you!