Akaflieg-Freiburg/enroute #91: Use CMAKE_FIND_ROOT_PATH when finding Qt5_TRANSLATION_DIR

EvilMav, 2020-10-11:
Minor build issue: in `buildscript-linux-debug.sh` `-DCMAKE_FIND_ROOT_PATH:STRING=$Qt5_DIR_LINUX` is passed to CMAKE, which would make you assume that passing a correct QT5 directory will make the build run, but `Qt5_TRANSLATION_DIR` is not being found in practice. On Arch, QT5 is under `/usr/share/qt`, but `CMakeLists.txt` under ` enroute/src/CMakeLists.txt ` says:
```
find_path(Qt5_TRANSLATION_DIR qtbase_de.qm
    HINTS ${Qt5Core_DIR}/../../../translations
    HINTS /usr/share/qt5/translations
    HINTS /usr/translations
    REQUIRED
    )
```

so if I pass `-DCMAKE_FIND_ROOT_PATH:STRING=/usr/share/qt`, find_path will be looking everywhere but where it is supposed to. Making it consistent would help development on different linux distributions.

PS Big thanks for your efforts on this project!

kebekus, 2020-10-12:
Dear EvilMav,

this has always been a headache. Please let me know if you have a concrete suggestion for improvement -- I will be happy to try it out on my machine(s).

Are you trying to package the Enroute App? If so, I'd be happy to put up a link on the homepage.

Also, we're constantly looking for competent people interested in helping with the project…

Best,

Stefan.

EvilMav, 2020-10-12:
Dear Stefan,

unfortunately, CMAKE and C++ aren't the things I'd call myself competent with at the moment - I'm mostly a python guy lately ^^ I've ended up patching `/usr/share/qt/translations` hint in which works for PKGBUILD, but there must be a more elegant way out there - after all CMake has facilities finding Qt5 packages if I read correctly...

I've only made a quick build script for AUR (Arch User Repository) for the current master branch: https://aur.archlinux.org/packages/enroute-git/. Seems to work =) I can add a stable release package at a later point.

I'd love to help. I'd be lying if I would say I can offer any significant amount of time at the moment, but may be some time later when PhD is coming to an end =) May I ask you if you are looking to provide more extensive flight planning capabilities similar to FL95, W&B, etc on desktop, or only concentrate on in-flight features?

Ilya

kebekus, 2020-11-07:
For lack of a better alternative, I added

- /usr/share/qt/translations

to the list of hints. I will then close the issue now.

closed 2020-11-07 by kebekus