Akaflieg-Freiburg/enroute #92: Add more documentation about where maps and other content changes can be made

LinuxKernel1, 2020-10-18:
In general there is some information about where the map content is coming from:
https://akaflieg-freiburg.github.io/enroute/maps/

Relevant information about the airports can be corrected on openaip.net but what about the airports traffic circuit patterns for example? Do I have to open a issue to get this kind of thing corrected or what are the necessary steps?

Please be more specific about where and how to make corrections about map contents and provided information.

kebekus, 2020-10-19:
Good point. I will look at this, but will take some time. Feel free to make concrete suggestions for improvement -- I am always open for help.

kebekus, 2020-12-01:
Implemented. The improvements will be contained in the upcoming release 2.1, that will likely appear in a few weeks.

Thank you for this helpful suggestion!

closed 2020-12-01 by kebekus