Akaflieg-Freiburg/enroute #104: More Privacy when Retrieving Weather Data

woodwind-enr, 2020-11-08:
Ein starkes Argument für Enroute ist m.E. "spioniert sie nicht aus" usw. Beim Wetter aber werden laut Warntext Standort und Strecke in die USA übertragen. Funktional ist das nicht nötig, und rechtlich finde ich hier

https://www.weather.gov/disclaimer

auch nichts dazu, sondern vielmehr

"The information on National Weather Service Web servers and Web sites is in the public domain, unless specifically annotated otherwise, and may be used freely by the public."

Dieser Disclaimer wird auch von der Flugwetterseite aus erreicht (https://aviationweather.gov/), die offenbar nur eine Unterseite der allgemeinen Wetterseite ist und woher ich problemlos diese Daten beziehe:

https://squawk-vfr.de/enroute/get_metar_taf.php?station=lfsb,edsb,edfm,edny,edds

Ich habe in meinem Webspace dieses kleine Programm hinterlegt, das METAR und TAF auf aviationweather.gov abgreift. Es könnte sicher auch an der Uni Freiburg laufen, braucht nur einen Webserver mit PHP und sehr geringe Ressourcen (Speicherplatz, Arbeitsspeicher, Rechenleistung, Bandbreite). Es kann eine Station oder mehrere auf einmal abfragen. Das Ausgabeformat lässt sich bei Bedarf noch anpassen. Aus meiner Sicht ist das legal und gleichzeitig frei von Verletzungen der Privatsphäre.

Die Dekodierung der Kürzel könnte wie bisher erfolgen; dafür existiert ja eine Mimik und inzwischen Langtexte in vier Sprachen.

Vielleicht wäre das eine Option zurück zu den hehren Grundsätzen der App.

woodwind-enr, 2020-11-08:
gib Nachricht, wenn du's auf Englisch brauchst

kebekus, 2020-11-08:
Language is fine -- we're all multilingual now, thanks to deepl.com

kebekus, 2020-12-20:
Hallo,

ich mag die Idee im Prinzip. Der vorgeschlagene Weg kann aber dazu führen, dass das "Aviation Weather Center" eine große Zahl von Anfragen von unserem Server sieht und dies für einen Angriff hält -- die meisten Services im Internet reagieren mit Alarm und sperren dann alle weiteren Anfragen von der betreffenden Adresse. 

Bevor man eine solche Lösung implementiert, sollte man also das das "Aviation Weather Center" informieren, um Einverständnis bitten und schauen, ob am eine kompetente und langfristig verbindliche Antwort bekommt (die Sache soll ja auch in einigen Jahren noch funktionieren). Hast Du Interesse, einmal nachzufragen?

Die "richtige" Lösung ist natürlich, das TOR-Netz zu verwenden. Der einbau eines TOR-Clients ist für mich allerdings im Moment vom Aufwand her nicht darstellbar…

Grüße

Stefan.

kebekus, 2023-01-31:
Hallo,

weil ich nicht noch weitere Infrastruktur betreiben möchte, habe ich mich für den umgekehrten Weg entschieden: die App zeigt am v2.24.0 beim ersten Start "Privacy Policies", die klarstellen, auf welche Seite zugegriffen wird. Danach ist kein kompliziertes Rechtemanagement mehr nötig. Das vereinfacht die Sache für mich sehr.

Grüße aus Freiburg,

Stefan.

closed 2023-01-31 by kebekus