Akaflieg-Freiburg/enroute #106: Custom geo-referenced maps

Helidude, 2020-11-18:
Would love to see a feature using own custom geo-referenced maps generated with [https://www.maptiler.com/](url)
Example: [https://www.youtube.com/watch?v=tdJgfuMKW1k](url)

kebekus, 2020-11-19:
Dear Helidude, 

can you  send me an example of a map file that you have in mind? Or better still, upload an example to this site?

Best, 

Stefan.

Helidude, 2020-11-19:
Hi
You could use any scanned paper chart or downloaded chart as pdf/jpg and convert it in Maptiler. 
Examples of free ICAO maps:
https://aim.naviair.dk/media/1161/edition-42-icao-anc-1_500000-denmark-pdf.pdf
https://www.austrocontrol.at/jart/prj3/ac/data/dokumente/icao_20_VS_254_rgb_geo_mR_LVR_2_2020-09-08_0709140.pdf

Subscriptions for digital maps are often expensive, and since I already own paper charts, it would be nice to be able to import those into the app. 
Maptiler exports in MBTiles format, but I see naviair.dk also have a already geo referenced map available for download in GEO-TIFF format. https://aim.naviair.dk/en/charts/

Would love to contribute to the project, but QT is unknown territory, coming from Java/C#
Best regards
Ivar

kebekus, 2020-11-22:
> You could use any scanned paper chart or downloaded chart as pdf/jpg and convert it in Maptiler.
> Examples of free ICAO maps: [...]

Understood. Can you please provide me with a few mbtile files that are produced this way? There are so many things to do that I am seriously grateful for any help that I can get.

Best,

Stefan.

kebekus, 2020-11-30:
Closed for lack of progress. Feel free to re-open once maps become available.

closed 2020-11-30 by kebekus