Akaflieg-Freiburg/enroute #111: Routings in CTR - data source unclear for me as user

spuckebaer, 2020-11-25:
Routings within a controlzone...remains unclear for me where you git this data from. Maybe to add source and issue date in documentation

kebekus, 2020-11-30:
Understood. I plan to implement a solution for the upcoming version 2.2.

kebekus, 2020-12-12:
Solved.

- The app homepage, https://akaflieg-freiburg.github.io/enroute/maps/#aeronautical-maps, now explains what part of the data comes from where.
- Version 2.2: The map management page of the app now has additional menu items for every map, "Map Info" that explains in detail where the data comes from and what revision of the data was used.
- Version 2.1: The bug report dialog has been greatly expanded and points the user to the proper place for reporting an error in the map data.

closed 2020-12-12 by kebekus