Akaflieg-Freiburg/enroute #112: Widen the frame of the airspace displayed

spuckebaer, 2020-11-25:
Use a Lenovo Tab10, quite huge. When flying specially in sunny condition a more wider frame (and/or higher contrast) might be usefull to ensure better visabilty

(Same for basic chart, higher contrast)

woodwind-enr, 2020-11-26:
using the maps suggested in #105 should deliver higher contrast

spuckebaer, 2020-12-01:
Not sure if #105 would solve this issue. Wide frame of airspace would be a first hige step. Maybe in the settings to choose as this might be different for small screens lile mobile phone and tablets.

I am quite happy with the basic map, quite simple and not that big - good for equipment with less memory. 
Anyhow: A change in contrast would be greatful

kebekus, 2020-12-12:
Solved. Increase line width be 40% for version 2.2.

closed 2020-12-12 by kebekus