Akaflieg-Freiburg/enroute #116: In case of unknown failures - sent logfile

spuckebaer, 2020-12-11:
In the recent days while playing a little bit around with enroute and charts transfer I had several failures and crashes.

Also im the past I had some minor issues I could not really see what went wrong.

Might be interesting for you to receive log files, right? If yes either in solution in APP or a procedure in handbook how to sent might by helpfull

kebekus, 2021-02-03:
Dear Spuckbaer,

to be very honest: I do not understand Android really well and I would not know how to implement your suggestion. Do you have some more insight here? Any help is highly welcome!

Best,

Stefan.

kebekus, 2022-01-16:
Dear Spuckbaer,

Google Play meanwhile does send useful stack traces that have helped me a lot. I will therefore close this issue. Thanks again for your suggestion!

Best,

Stefan.

closed 2022-01-16 by kebekus