Akaflieg-Freiburg/enroute #117: Night Modus

propellertje, 2020-12-11:
Eyes are rather sensitive to light especially in a dark environment such as Night Flying.

Screen devices are rather bright and sometimes it is not sufficient to reduce the brightness of a device for Night Flying.

Request is to impleent a night modus. Kind of a Dark Modus.
See the screen of the commercial product Foreflight.

Keep up the good work! Kind regards Bernard

kebekus, 2020-12-12:
Thank you for your suggestion. I am considering this seriously, but I cannot promise anything at this stage. Perhaps you could attach a screenshot of foreflight? -- Thanks.

propellertje, 2020-12-24:
https://foreflight-www.s3.amazonaws.com/assets/weather_double_chrome_www.png

Dear sir, Sorry for my late answer. Hope this helps. Kind regards Bernard

kebekus, 2020-12-25:
Thanks for the image. I will look at this, but please give me a little time.

propellertje, 2020-12-25:
You welcome, no hurry and thanks a lot for your effort!

kebekus, 2021-01-22:
Done. The feature will appear in v2.3.0, which is scheduled to appear in mid-February. Thanks again for the suggestion!

closed 2021-01-22 by kebekus

propellertje, 2021-02-02:
Thank you so much! Really appreciate your effort. Kind regards, Bernard