Akaflieg-Freiburg/enroute #122: Linux - compile - cannot run

slawekmikula, 2020-12-20:
After compiling by myself I cannot run enroute on linux box.

source: latest git (fab7e264)
ubuntu 18.04 
Cmake: 3.19.2 (custom dir)
Clang: 10 (os package)
Qt: 5.15.2 (custom dir)

Compilation without issue. When running I got error as in the attachment. [error.txt](https://github.com/Akaflieg-Freiburg/enroute/files/5721383/error.txt)

For some time there is visible main window (1-2s) but without any possibility to interact:

[image]

kebekus, 2020-12-21:
Dear slawekmikula,

thank you for your report. The issue that you report is a known issue in Qt. I reported the issue in June '20, but the Qt company does not seem interested in bugfixing.

- https://bugreports.qt.io/browse/QTBUG-84768

Workaround: do not compile the app with the address/memory sanitizer, then it will run just fine.

Any help with fixing this bug is highly appreciated. 

- If you have the time, it might help if you could go to the Qt bug report, confirm that you see this issue, and ask for a fix.
- Please contact me a stefan.kebekus@gmail.com if you want to help (or if you want to help with other aspects of app development)

I am closing this bug report now, because I do not know what else to do.

Best,

Stefan.

closed 2020-12-21 by kebekus

slawekmikula, 2020-12-21:
Ok, I understand but how to build without address/memory sanitizer ? Maybe if this is error still persistent you can modify the linux build script in the root folder to not use it by default.

I've removed --sanitize options from the build commandline but it still breaks after run. Sorry new to CLANG :)

kebekus, 2020-12-21:
Just use GCC

makedir build
cd build
cmake ..
make

slawekmikula, 2020-12-21:
Sorry for the last noise. I haven't removed sanitizing from linker. When it is removed all went OK and application runs. 

Maybe for future:
- give some information about this issue in the readme/build info 
- remove sanitizing from linuxbuild shell script - it would be better for anyone who want to build it first time :)