Akaflieg-Freiburg/enroute #124: can not save flight routes

zehess, 2020-12-25:
Error Message: 
"unable to open the file 'storage/emulated/0/enroute flight navigation/flight routes/...' geojson for writing"
Android 10
moto g(7) plus

kebekus, 2020-12-28:
Thank you for your report. So far, I was unable to reproduce the issue. Can you please provide me with more information (what file name did you choose?). Perhaps you can also attach a few screenshots or a screencast video that shows the issue?

Best,

Stefan.

zehess, 2020-12-28:
Hi Stefan,

danke für die rasche Nachfrage!
Es geht bei mir um das erstmalige Speichern einer Route über
"Flugstrecke"-"in Bibliothek speichern", Speichername wie vorgeschlagen
(funktioniert aber auch mit eigenem einfacheren Namen nicht). Screenshots
anbei.

Der In der Fehlermeldung erwähnte Dateipfad ist vorhanden und beschreibbar.

Vielen Dank für die Mühe
Carsten Schneider

kebekus, 2020-12-29:
I am able to reproduce the issue now. Currently working on a fix.

kebekus, 2020-12-30:
Hallo,

ich glaube, dass ich das Problem gelöst habe. Kannst Du bitte die aktuelle Beta-Version (=2.2.0) der App installieren und bestätigen, dass das Problem nicht mehr auftritt? Um die Beta-Version zu bekommen, musst Du dich über den folgenden Links als Beta-Tester anmelden:

    https://play.google.com/apps/testing/de.akaflieg_freiburg.enroute

Alternativ kannst Du Dich in "Google Play" als Beta-Tester anmelden. Du kannst das Beta-Programm natürlich jederzeit wieder verlassen und zur stabilen Version der App zurückkehren.

Grüße,

Stefan.

zehess, 2020-12-30:
Hallo,

ja, Flugstrecken speichern, laden und umbenennen funktioniert jetzt
tadellos.

Vielen Dank und ein schönes 2021

C. Schneider

kebekus, 2020-12-30:
Wunderbar, danke für den Bug Report!

closed 2020-12-30 by kebekus