Akaflieg-Freiburg/enroute #125: Strings only in original language - cannot translate

slawekmikula, 2020-12-27:
Some not translated strings i've found during PL translation (pushed to the enrouteText repository).

[image]
[image]

kebekus, 2020-12-28:
Thank you!

- I added translations for the "Total: …" text.
- The License text is found in Librarian.cpp and can be translated. As of now, the Polish translation is however a copy of the English text. A attach a screenshot of the Qt Linguist

[image]

closed 2020-12-28 by kebekus

slawekmikula, 2020-12-28:
Ok, i'll translate it (it must be translated -> copied) before so it does not display in Qt Linguist as untranslated