Akaflieg-Freiburg/enroute #127: Programm lässt sich auf Lenovo-Tablet mit android 7 nicht runterladen

knispe, 2020-12-30:
Das Programm lässt sich über den playstore nicht auf ein Lenovo-Tablet (Android 7) runterladen. Ist ein neueres Android erforderlich? 
Beste Grüße

kebekus, 2020-12-30:
Wir fordern im Moment nur Android 5. Ich sehe, dass Sie den Report zwischenzeitlich geschlossen hatten, hat sich das Problem inzwischen vielleicht gelöst?

knispe, 2020-12-30:
Leider nein, das Schließen erfolgte versehentlich. --Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

kebekus, 2020-12-30:
Mir ist unklar, ob ich hier helfen kann. Auf jeden Fall benötige ich folgendes:

- Genaue Bezeichnung des Gerätes (evtl. Screenshot von der betreffenden Seite der Einstellungen)
- Genaue Beschreibung der verwendeten Android-Version (auf jeden Fall Screenshot von der betreffenden Seite der Einstellungen)
- Genaue Beschreibung, wie das Herunterladen fehlschlägt, ebenfalls am besten mit Screenshots, die das Problem zeigen.

Sie können die Screenshots hier auf GitHub direkt hochladen.

knispe, 2020-12-30:
Vielen Dank für Ihre Mühe,

 

ich denke das Problem hat sich erstmal erledigt .

 
 

Gesendet: Mittwoch, 30. Dezember 2020 um 11:27 Uhr

kebekus, 2020-12-30:
Problem laut user erledigt.

closed 2020-12-30 by kebekus