Akaflieg-Freiburg/enroute #129: User feedback: Route Import failed

woodwind-enr, 2021-01-03:
User input:
I Took three routes from Sky Demon and Exportet to .gpx
When importing to Enroute, got a Error Message. See Screen Shot attached.
Reimporting .gpx to Sky Demon works fine. 

[image]

kebekus, 2021-01-04:
Thank you! Can you please upload the relevant GPX file here, so that I can have a look at it?

kebekus, 2021-01-04:
See sample GPX files below. These files do look correct, so there might indeed be a bug in Enroute. I will investigate.

[Routen.zip](https://github.com/Akaflieg-Freiburg/enroute/files/5766513/Routen.zip)

kebekus, 2021-01-06:
I can reproduce the issue now; this is a problem with the MIME database in Android, so the file type is not correctly identified. Currently working on a fix.

kebekus, 2021-01-07:
I just uploaded version 2.2.3 to Google Play as a beta version; this version (hopefully!) fixes the bug. Version 2.2.3 will be distributed later today.

@woodwind-enr Do you have a chance to install the beta and confirm that the problem is gone? -- Thanks.

woodwind-enr, 2021-01-07:
[]()will try, probably today

woodwind-enr, 2021-01-07:
1. works! 
2. says 'waypoint' instead of using the names out of the file
3. same green line as fis sectors (?), should contrast better to other lines
4. restart of app in normal way tried to load temp file again but did not find, of course
Can do more later today...

kebekus, 2021-01-08:
Fixed in version 2.2.3, which will be distributed by Google Play today.

Thank you for the report!

closed 2021-01-08 by kebekus