Akaflieg-Freiburg/enroute #130: Error saving route

a533, 2021-01-04:
Hi,

Excellent app. I get an error saving a route

An error occurred...
Unable to open the file '/storage/
emulated/0/enroute flight
navigation/flight routes/LECU-
LEAS.geojson' for writing.
OK

I have Android 10 and storage permission granted.

Thanks and regards

kebekus, 2021-01-05:
Thank you for your report. The issue is fixed in the upcoming release 2.2.2, which I upload to Google Play today.

kebekus, 2021-01-05:
Please feel free to re-open this issue in case that the problem persists after the update.

closed 2021-01-05 by kebekus

a533, 2021-01-05:
Hi Stefan, it's a great app that you have made. I like it very much. I am a
developer also. These are my apps.

https://play.google.com/store/apps/details?id=com.qrouting

https://play.google.com/store/apps/details?id=com.AviationChecklist

www.qrouting.com

I am a commercial pilot with instructor rating and IFR rating I fly in
Madrid, LECU, and as you see my app is mostly for planning and weight and
balance. I lack  the moving map part but I connect my app with oruxmaps (
https://www.oruxmaps.com/) I would like to participate in the development
of your app if you want me to.

Regards

Alvaro