Akaflieg-Freiburg/enroute #150: Display bug

acrovato, 2021-01-24:
Hi Stefan,

André, a fellow Belgian pilot, reported a display bug. All the contextual text is in capital letter and displayed in a weird light font (see attached screenshot). I tried to reproduce the bug on my side, but did not manage (which is a kind of good news I guess).

According to André, the app was working normally up to this week, and he did not make any changes to his phone.
If I had to make a guess, I would say that the issue come from an update linked to Android, or something similar...

Here are the device information:
Device: One+ Nord 
OS: Android 10-5-10 Oxygen OS

[image]

kebekus, 2021-01-25:
Hi,

I have looked at this and I am totally clueless. I have **no idea at all**. I have seen some devices with bitrot, which seems to happen much more often than thought. If bitrot is the problem, then uninstalling and re-installing the app will help. If this is a matter of Android fonts/configuration, then uninstalling and re-installing should not have any effect. Can you please ask your user to try this out? -- Thanks!

Best,

Stefan.

andre-ada, 2021-01-25:
Hello friends,
Thank you for the message.
I followed the procedure suggested by Adrien and then I reinstalled Enroute. 
The presentation is unchanged: the letters are written in upper case and the data is barely readable (see previous screenshot).
The smartphone is very recent and all other apps are working fine.
Fortunately, everything else in Enroute works perfectly.
Thank a lot for your attention.
André

kebekus, 2021-01-26:
Dear Andre-Ada,

thank you for your quick reply -- there go my hopes for a quick solution. I plan to release the next beta version later this week. Could you subscribe to the beta chanel on Google Play and try it out? The next version comes with a bit of a technology upgrade. Maybe that helps your problem…

Best,

Stefan.

andre-ada, 2021-01-26:
Hello Stefan,

Thanks for your message.
I'm now beta.
Andre

kebekus, 2021-01-29:
Have you had a chance to look at the new beta? Does the problem persist?

andre-ada, 2021-01-29:
Hello Stefan,

Thank you for the message.

Update: no change in the display of texts, always in capital letters (see
screenshot).

Fortunately, this does not affect the correct functioning of the program.

For my part, I also asked the manufacturer of my recent OnePlus Nord
smartphone. This problem is unknown to them.

Thank you for your dedication to improving Enroute.

André

andre-ada, 2021-03-09:
Hello Stefan, bonjour Adrien,

I just updated my smartphone with Androïd 11.
Good news! No more problem with the capital letters.
See annexed screenshot.
Have a nice day.
André

kebekus, 2021-03-09:
Thanks for all your help. To be honest, I still have no idea what's going on … I close the issue nonetheless.

closed 2021-03-09 by kebekus