Akaflieg-Freiburg/enroute #154: Screenshots for F-Droid

mondlicht-und-sterne, 2021-01-29:
You can add Screenshots for F-Droid, too, if you want.
https://f-droid.org/en/docs/All_About_Descriptions_Graphics_and_Screenshots/
https://gitlab.com/snippets/1895688
That was simply information about what is still possible. You would then only have to enter the
screenshots in fastlane. This is just one way that you can also add screenshots to the app view
if you can. If you added screenshots (then F-Droid is happy) and if you can't add any -> then
you can close the issue because it was only an info. Thank you that your app is available in
fdroid. Very much for your attention.

kebekus, 2021-02-03:
Thanks for letting me know. We're working on this.

kebekus, 2021-05-04:
Added Fastlane structure as requested, to provide screenshots on f-droid.

Thank you for your help!

closed 2021-05-04 by kebekus