Akaflieg-Freiburg/enroute #156: IP change in settings for Wifi connected devices like Stratux

spuckebaer, 2021-02-21:
Connect the Stratux to enroute is workung well.
But it is needed to adapt the IP in Stratux. 
As it is more likely to move the tablet or smartphone arround the different planes it would be perfect to allow IP settings in Stratux

kebekus, 2021-02-22:
@spuckebaer Thanks! I will look at this and try to find a solution that works automatically. This is however not my top priority at the moment (assuming that people who use Stratux are able to change the IP), so it might take a while before a solution is published.

spuckebaer, 2021-02-22:
Thanks....very clear. not really mandatory but nice to have

kebekus, 2021-04-07:
Starting with 2.6.0, the app will check several IP-Addresses simultaneously. Changing the IP address of the Stratux device is then no longer necessary. Please have a look at the current beta to try this feature.

closed 2021-04-07 by kebekus