Akaflieg-Freiburg/enroute #157: Enroute Crash when Stratux connected

spuckebaer, 2021-02-21:
When Wifi connection to Stratux is established and Enroute is started is crashes. Restart of Enroute is solving the problem. (This is systematic at each start)
+ During yesterdays flight connection was 1 time lost even if wifi was connected

No failure message, just saw connection in Information "Trying to connect"

Lenovo Tablet lenovo tab m10 tb-x505f
Android 10

kebekus, 2021-02-22:
@spuckebaer Thanks for the message! I will look at this with top priority. Regretfully, I have no idea at present why the app would crash, and any additional piece of information would help (screencast video, …). How well do you know android? Can you look at the log files and check if there is any useful information in there?

spuckebaer, 2021-02-22:
I tried ADB but could not manage...if there are any proposals how to grab info from my tablet I am happy to provide.
I will also check tonight how I can make a video

spuckebaer, 2021-02-23:
I habe currently a Power supply issue with my stratux and therefore I cannot reproduce the topic. If wifi is not connected to stratux start procedure is as normal without crash

spuckebaer, 2021-03-04:
Still waiting for spareparts.....

kebekus, 2021-03-05:
This problem still puzzles me. Let's wait for your spare parts and see…

kebekus, 2021-04-07:
Dear spuckebaer,

can you please check if the problem still shows with the latest beta version of the app?

Thanks,

Stefan

spuckebaer, 2021-04-07:
Still not received the new parts...friend of mine has a similar setup and it works for hinm,...so I concider it is on my side.Would propose to close and when I got the new parts I will check again and let you know if still the same, o.k.?

kebekus, 2021-04-08:
Ok, will proceed as you suggest. Thanks for your help!

closed 2021-04-08 by kebekus