Akaflieg-Freiburg/enroute #158: Green Target Arrow optical to similar to "Naturschutzgebiete"

spuckebaer, 2021-02-21:
Maybe to think about to change the colour of the arrow to ensure better visabiliy .
Fesable to add a flight path vector for the next minute (Due to direction and speed)?

kebekus, 2021-02-22:
@spuckebaer I will look at this. It might take a while because some other issues have a higher priority at the moment.

kebekus, 2021-02-22:
Kannst Du mir einen Screenshot schicken?

spuckebaer, 2021-02-22:
[image]
Hab gerade kein Screenshoot aus der echten Skalierung beim fliegen, aber glaub für eine grobe Idee könnte es vielleicht schon reichen. Falls du mehr brauchst kann ich gerne am Freitag beim fliegen mehr machen

kebekus, 2021-04-08:
Das Problem sollte sich mit der Anpassung der NRA-Farbe gelöst haben…

closed 2021-04-08 by kebekus