Akaflieg-Freiburg/enroute #159: Own hex code of transponder/flarm

spuckebaer, 2021-02-21:
When flying enroute displays a green target arrow follow you all times. (Own aircraft :-))
Is it feasable to add in enroute setting a hex code not to be shown on display?
In best case to add add several code for pre selection, so that in case switching aircarft I can easily select the aircraft I am just flying

kebekus, 2021-02-22:
@spuckebaer Can you confirm: you are flying with **two** traffic receivers in one plane?

spuckebaer, 2021-02-22:
I flown with our Piper with just a Mode S transponder and used a stratux with OGN and ADS-B receiver. So 1 signal was received from mode S on stratux.
I may have the chance to fly one of our Aquila on friday with Mode S AND Flarm, so I assume I will be followed by 2 targets :-)

b3nn0, 2021-03-09:
Just stumbled upon this.
You can configure multiple Codes in the Stratux Web Interface under Settings->Mode S Code.
Stratux will automatically try to determine which of the entered codes you are actually flying with, so you will still receive warnings for other aircraft, even if you entered the code in there. But it will filter out your own aircraft.

kebekus, 2021-04-07:
In view of b3nn0's comment, I guess that the issue can be closed then?

closed 2021-04-07 by kebekus