Akaflieg-Freiburg/enroute #160: Contrast of "Naturschutzgebiete"

spuckebaer, 2021-02-21:
I like very much the tnew contrast of the maps. Only the NSG green collour is a little bit to heavy for me and it takes my attention away from the other essential features. Would be nice to adapt

kebekus, 2021-02-22:
I will look at this. It might take a while because some other issues have a higher priority at the moment.

spuckebaer, 2021-02-22:
Perfect, THX

kebekus, 2021-04-08:
Thanks again! Following your suggestion, I made the color a little more subtle for the next app version. I have to walk a fine line here because NRAs are extremely important in some countries, such as Austria, where there are many of them and violations are punished with extreme fines (~10k€).

Best

Stefan.

closed 2021-04-08 by kebekus