Akaflieg-Freiburg/enroute #161: clang 10 - error include <compare>

slawekmikula, 2021-02-21:
Current head. Building using ubuntu 18.04, (clang version 10.0.0-4ubuntu1~18.04.2)
Error from commit: d30155a7

In file included from src/CMakeFiles/enroute.dir/Unity/unity_1_cxx.cxx:3:
In file included from src/FlightRoute_GPX.cpp:24:
In file included from ../src/FlightRoute.h:30:
In file included from ../src/Aircraft.h:24:
../src/AviationUnits.h:23:10: fatal error: 'compare' file not found
#include <compare>
         ^~~~~~~~~
1 error generated.

GeorgVFR, 2021-02-21:
I had a similar issue with ubuntu 20.04. [libstdc++-10-dev](https://packages.ubuntu.com/groovy/libstdc++-10-dev) was missing.

kebekus, 2021-02-22:
@GeorgVFR Thank you!

@slawekmikula Can you confirm that missing libstdc++-10-dev was the issue

slawekmikula, 2021-02-22:
Nope, ubuntu 18.04 have only https://packages.ubuntu.com/bionic/libstdc++-7-dev and in it, there is no compare algorithm header/implementation. I think with gcc 7.5 (current on 18.04) it cannot be achieved. Ubuntu 18.04 cannot right now compile it with stock packages (AFAIK).

kebekus, 2021-02-23:
Annoying. I will try to remove the C++20 code then (though it *is* convenient)

slawekmikula, 2021-02-23:
It depends what is minimal gcc/clang/OS version to compile enroute on Linux. Ubuntu 20.04 can compile it. 

Unfortunately (haven't got time lately to dig better) there is no easy PPA (custom deb repository) for Ubuntu 18.04 with more recent gcc/g++ libraries.

kebekus, 2021-02-23:
Fixed in the master branch, should compile just fine now.

closed 2021-02-23 by kebekus

slawekmikula, 2021-02-23:
Include is still there. Needs fix.

Removed it and all went well. Thanks :)

kebekus, 2021-02-23:
Oops. Fixed.