Akaflieg-Freiburg/enroute #167: Route export to an other device

propellertje, 2021-03-15:
**Describe the bug**
Plan a Route on device "A" and save it as a GPX file.
Open device "B" and open the saved GPX file.
Enroute will start but the route will not display on device "B".

**To Reproduce**
- Enroute Flight Navigation 2.5.0 (both devices)
- After having planned a Route
- Share ...
- ... to GPX file
- select Drive Save to Drive
- confirm filename.gpx, email account and folder, click save
- download from Drive on an other device
- select from that device filename.gpx
- download filename.gpx
- via download folder select filename.gpx 
- enroute will start but the "filename.gpx" route does not show up.

**Expected behavior**
I do expect to see the planned Route from device "A" to show up on device "B".

**Screenshots**
[FROM device "A"] The route to be shared across another device
[image]
[TO device "B"] The route to be received after having received the GPX file which opens Enroute Navigation
[image]

**Smartphone (please complete the following information):**
DEVICE "A"
 - Device: [Lenovo TB8705F]
 - OS: [Android 10]

DEVICE "B"
 - Device: [Honor 9 STF-L09]
 - OS: [Android 9]

**Additional context**
Thanks for helping out, kind regards Bernard

kebekus, 2021-03-15:
Thanks for your report. Can you attach the GPX file, please?

propellertje, 2021-03-15:
Hello Stefan,

The file you asked for.

Kind regards, Bernard

kebekus, 2021-03-16:
Dear Bernard,

somehow the file did not make it to GitHub. Perhaps you could send it to me by mail? I tried with my own files and was unable so far to reproduce the issue…

Best -- Thanks for your help,

Stefan.

kebekus, 2021-03-16:
Nope. It seems that GitHub does not allow GPX files as attachments to Mails. Maybe send a mail to me directly? -- Thanks.

propellertje, 2021-03-17:
On device "B" I did noticed that my GPS location was disabled. I enabled GPS and I was able to sucessfully import the GPX file. It seems that the application works fine and that it is important to have the GPS location enabled beforehand. I will do some further testing on my other Android devices and will give an update. Kind regards, Bernard

propellertje, 2021-03-17:
I also enabled the GPS location on en Huawei tablet M5 and it is working perfectly! As far I am concerned this issue is resolveld! Kind regards, Bernard

propellertje, 2021-03-17:
Trigger was the message from Stefan that he was unable to reproduce the error. That brought me to investigate the issue further on my side. Bottom line for sharing Routes do enable the GPS. May be you can confirm. Kind regards, Bernard

kebekus, 2021-03-17:
Thanks for the investigation. This is helpful indeed. I will further look into the issue.

kebekus, 2021-03-18:
Hmm. I found that I still cannot reproduce the issue. On a test-system with Android 9, the route import worked fine regardless of whether GPS was enabled or not.

- Can you please check if you can reproduce the issue on your Android 10 device?
- Would it be possible that you take a little screencast video so that I can see how the app behaves? This has often been helpful in the past.

Thanks - and best wishes,

Stefan

kebekus, 2021-04-08:
Dear Bernard,

did I ever say "thank you" for the screencast file? Thank you!

I am still unsure about the nature of the problem, but I do have a guess now. Could you please try the latest beta version of the app (=2.5.102, to appear on Google Play later today) and tell me if the problem is still there? That would be extremely helpful.

Best

Stefan.

propellertje, 2021-04-08:
Hi Stefan, you welcome and it works!!! I did upgrade both devices: Android
phone and the Huawei M5 tablet. After the export on the M5 I was able to
import the route on the honor 9. Excellent work. 👍😀 Thank you so much for
your time and effort. Keep up the good work. Kind regards Bernard

kebekus, 2021-04-09:
Thank you for the feedback. The issue seems to be solved, I close it now.

closed 2021-04-09 by kebekus