Akaflieg-Freiburg/enroute #168: Skyecho2

jandrewartha, 2021-03-18:
The app will not connect to Skyecho2. The connect button greys out for a few seconds then returns to normal. But there is still no connection.

kebekus, 2021-03-19:
Hi,

thanks for your report! I have never seen a Skyecho2 device, but I had a brief look at the specs. The Skyecho2 sends out data in "GDL-90" format, which is currently not supported by the app. I could probably implement GDL-90 in the coming months if there is sufficient interest. Hence my question: Do you have access to a Skyecho2 device and would you be willing to run beta-versions of the app to test the feature?

Best wishes,

Stefan.

jandrewartha, 2021-03-19:
Hi Stefan,

Thanks for looking at this issue. I am a glider pilot in the UK, currently
not flying due to Covid restrictions. I have recently bought a Skyecho2 to
add ADSB to my glider.
I have been looking for an EFB app which can display ADSB In from the
Skyecho2 and came across Enroute Flight Navigation. So far I have been
impressed by the app but I have yet to fy with it.

I am surprised that it does not recognise the GDL90 protocol as this seems
to be a common standard for EFBs and Traffic devices such as Stratux,
Skyecho2, etc. I couldn't find any information on your website or the
manual about what protocols it does support. In the UK the CAA has been
providing 50% grants for GA pilots to buy Electronic Conspicuity devices so
there has been a big increase in the number of aircraft equipped with them.
Skyecho2 has been the most popular device as it is relatively cheap for
what it provides.

I am very willing to try beta software to test a GDL90 interface if you
implement it.

Best regards,
Joe

kebekus, 2021-03-22:
Dear Joe,

thanks for your reply. I will implement GDL90, but this will take some time -- I develop the app alone and in my spare time. I will come back to you as soon as I have a beta version that is ready to test.

Here in Germany, things are different. Pretty much all flight clubs installed FLARM devices in their aircraft some 20 years ago. I have therefore never seen anything else … all the devices that I came across speak FLARM NMEA.

Best,

Stefan.

jandrewartha, 2021-03-22:
Hi Stefan,

Thanks for agreeing to implement GDL 90. I understand that it will take a
while.

In the UK nearly all gliders have Flarm but it is not so popular with
powered aircraft. The UK CAA want all aircraft to have Electronic
Conspicuity and ADSB is their preferred solution. Hence the 50% grants to
encourage people to fit EC.

I await your beta.
Joe

kebekus, 2021-05-05:
Dear Joe,

the latest beta-version of "Enroute Flight Navigation" (=2.6.9) **should** be able to work with your SkyEcho2 device, but I have never tested it with an actual SkyEcho device. The changes to the program were pretty massive, with the obvious potential of introducing new errors. I would greatly appreciate it if you could test the new functionality thoroughly.

- Does the app connect reliably to the device? Is the connection sufficiently robust for practical use?
- Is the device position properly displayed in the moving map? Same for horizontal speed.
- Are geometric height and pressure height shown correctly?
- Does the app show the traffic correctly in the moving map? Are positions, directions, vertical distances, and vertical speed of the traffic shown correctly?

Thanks again for your support!

Stefan.

jandrewartha, 2021-05-05:
Hi Stefan,

Thanks for doing all this work. I have installed 2.6.9 on my phone. I can't
try it at home as the Skyecho is installed in my glider. I will check it
out when I next fly and report back. This may take a little while depending
on suitable weather.

Best regards,
Joe

jandrewartha, 2021-05-20:
Hi Stefan,

I have not been able to check it in the air but I have tried it on the
ground. It connected to Skyecho ok and position was correct. I was able to
see traffic in the area and as far as I could tell the position, height,
etc seemed correct. I shall report back when I've tried it in the air. The
weather here has not been good recently for flying gliders.

Thanks again for the great work. So far I'm impressed.

Joe

kebekus, 2021-05-21:
Thank! I do hope that the weather will improve, and I am looking forward to your full report!

Stefan.

kebekus, 2021-06-14:
Dear Joe,

have you had a chance to look at this?

Best

Stefan.

jandrewartha, 2021-06-16:
Hi Stefan,

Unfortunately not. My aircraft has been in the workshop for the last few
weeks. And at the moment I am in Thailand visiting my son and his family
for the first time since the pandemic started. I hope to be flying again in
August and I should be able to evaluate the app better then.

Best regards,
Joe

kebekus, 2021-07-09:
Users report that the SkyEcho2 is working fine. I am closing this issue and thank everyone involved!

Best

Stefan.

closed 2021-07-09 by kebekus