Akaflieg-Freiburg/enroute #171: GPS Waypoint coordinates

propellertje, 2021-04-01:
Adding a Waypoint is a great Feature. 
Now it is also possible to give a Waypoint a name defined by the user.
My request would be, if possible, to implement the GPS coordinates of a Waypoint that could be edited.
Why? Because sometimes adding a waypoint is not that precise.
By adding and making GPS coordinates Editable like the Waypoint name would be a great new feature.
[image]
GPS notation like DFS:
[image]

The first screenshot is from MACH7 (https://appli.mach7.com/)
The second screenshot is from (https://secais.dfs.de/pilotservice/user/login/login_edit.jsp)
Both require a free registration to access.

Kind regards, Bernard

propellertje, 2021-04-01:
It is a request, not an issue of course ... ;-)

kebekus, 2021-04-01:
Thanks for the suggestion. The feature is on my TODO-List. However, I am programming this app alone and in my spare time, so it will take a long while before the feature will be implemented. Any help is welcome!

propellertje, 2021-04-02:
I understand, no problem! Thank you for considering this enhencement in the future. Kind regards, Bernard

propellertje, 2021-04-02:
https://www.movable-type.co.uk/scripts/latlong.html
Link with example for lat/long/bearing theory and scripts reference in C++ and other languages.
Maybe this will help a little bit... Kind regards Bernard