Akaflieg-Freiburg/enroute #178: Connect to flight simulator X-Plane 10

propellertje, 2021-05-04:
Nice feature to connect to a flight simulator.

I did try EFN version 2.6.9 with X-Plane 10 but I am out of luck.

Alternative is the X-plane to GPS app but is not supported any more in the Google Play Store.
The latter is still working on a mediapd M5 10.8" and Honor 9, not on a Lenovo tab TB-8705F.
You also need to spoof the GPS location and must therfore bring the Android device in developper mode
[image]

PC (X-Plane 10) is connected on the same WIFI network as the Android device.
See picture

It's a nice to have feature and hope this feedback is helpful.

Kind regards, Bernard

GeorgVFR, 2021-05-05:
Hi Bernard,
I just tested EFN with X-Plane 10 successfully including symbols for nearby traffic.

[image]

Instead of enabling data output on tab "Data" please go to tab "iPhone/iPad" and enable "send AHRS data to ForeFlight, WingX Pro, or SkyDemon on ALL iPads or iPhones". Please be sure to disable output of data on tab "Data" before.

[image]

A description how to connect X-Plane 10 to ForeFlight can be found here:
https://www.x-plane.com/2012/08/foreflight-charts-supported-in-x-plane-10-10-beta-9/

Regards,
Georg

kebekus, 2021-05-06:
Thank you. I will close this issue now.

@GeorgVFR I guess you agree that I copy your text (and screenshot) to the manual?

closed 2021-05-06 by kebekus

kebekus, 2021-05-06:
@GeorgVFR  Did I say "Thank you"? -- Thank you!

GeorgVFR, 2021-05-06:
Thank you for this wonderful app!

> I guess you agree that I copy your text (and screenshot) to the manual?

Sure, please use everything you want.

propellertje, 2021-05-11:
Hi Stefan and Geaorg, thank you so much! It works! on Android device. Kind regards, Bernard
[image]

propellertje, 2021-05-19:
Hi Georg and Stefan, Thank you so much!!! It works!