Akaflieg-Freiburg/enroute #184: [v2.7.2] Bugs with route library and waypoints

acrovato, 2021-06-13:
I have recently updated enroute to v2.7.2 (commit ce1bf5bf61c6a19bb61108711075201d28d45378), and I have noticed two minor bugs with route library and waypoints.

**Route library**  
Clicking on route, or opening it (by using the horizontal three dots, then "Open..."), from the "Route library" menu DOES NOT load the route. Removing, renaming, sharing and opening the route in another app work as expected.

*Waypoints**  
When removing a waypoint from the moving map, by selecting the waypoint and pressing the "-" sign, that waypoint IS NOT removed from the route, but the FIRST waypoint of the route is ALWAYS removed instead. Manipulating the waypoints from the "Route" menu works as expected.

kebekus, 2021-06-14:
Thank you for the report, this is extremely helpful! Recently, I had to re-write large parts of the code...

kebekus, 2021-06-19:
@acrovato Dear Adrien, can you confirm that the issue is solved in the current beta version (=2.7.3)? -- Thanks!

acrovato, 2021-06-20:
Hi Stefan,

Both reported bugs seem corrected and the app now works as expected.

Good job and thanks for your dedication!

closed 2021-06-20 by acrovato