Akaflieg-Freiburg/enroute #186: Adding Prohibited, Restricted and Danger areas on the map

propellertje, 2021-06-22:
Would be helpful to make the map more complete for the NL by adding the Prohibited, Restricted and Danger areas.
eAIP NL: https://www.lvnl.nl/eaip/2021-06-03-AIRAC/html/index-en-GB.html
section ENR 5 Navigation Warnings, 5.1 ---> ENR 5.1 PROHIBITED, RESTRICTED AND DANGER AREAS
Reason if an area becomes active via NOTAM notification I can plan the route accordingly taking account of the restriction.
During flight it helps me to avoid those active areas.
The graphical map can be found on the eAIP NL (Enroute Chart - ICAO) ENR 5 Navigation Warnings ---> ENR 6

**Describe alternatives you've considered**
I did consider the app https://vfrnav.com/ but EFN offer better features for my personal needs.

**Additional context**
Scrrenshot ENR 6 Enroute Chart - ICAO
[image]

Thank you for considereing this request.

Kind regards, Bernard

kebekus, 2021-06-23:
Dear Bernard,

I am confused, because it seems to me that danger areas, restricted areas, and prohibited areas **are** shown. See the screenshots attached. Can you clarify the issue? Are there additional areas that you want to see? Do you have an example?

Best,

Stefan.

[image]
[image]
[image]

propellertje, 2021-06-23:
Hello Stefan, Thanks for the fast response. Indeed you are right. Some areas are mentioned but not all. As an Example from my homefield EHGG due north some areas are missing. Herby enclosed one screen shot from EFR 2.8.0 and on from the orginal map NL 2021. The ENR tables mention all the areas.
[image]
[image]

I hope this will clarify the mentioned issue.

Thank you for looking to this.

Kind regards, Bernard

kebekus, 2021-06-24:
Am 23.06.21 um 09:12 schrieb propellertje:
>
> Hello Stefan, Thanks for the fast response. Indeed you are right. Some areas are mentioned but not all. As an Example from my homefield EHGG due north some areas are missing. Herby enclosed one screen shot from EFR 2.8.0 and on from the orginal map NL 2021. The ENR tables mention all the areas.

Dear Bernard,

I understand. This issue cannot be solved by the app, because the app simply displays air space data from the public openAIP database -- and the areas that you mention are simply not present in openAIP.

openAIP is used by many flight navigation systems. I believe that the data should absolutely be added there, and I would like to ask you to turn to the people at openAIP.net and discuss the matter with them. Peter Kemme <peter.kemme@openflightmaps.org> (in CC) knows openAIP well and has kindly volunteered to help with these issues; it would probably be a good idea to contact him first.

@Peter: Bernard pointed out that a number of (restricted|prohibited|danger) areas are missing in openAIP for the Netherlands. For examples, see the many restricted areas north of Groningen (including the EDH-R2), which are present in openFlightMaps, but not in openAIP. I attach screenshots. Would it be possible to add those? It seems that many (maybe all?) of the missing areas are "activation by NOTAM" … is there a policy at openAIP to not include those? -- Any help is greatly appreciated!

Best,

Stefan.

PS: I am closing this issue, because this cannot by solved by the app…

[image]
[image]

closed 2021-06-24 by kebekus

propellertje, 2021-06-25:
Dear Stefan, thank you very much for you're elaborate explanation and help
to resolve this issue! Kind regards, Bernard