Akaflieg-Freiburg/enroute #187: Possibility to add custom content

manuuuu400, 2021-06-24:
It would be a great addition for the app to support custom base maps. I have a couple of mbtiles of current ICAO VFR charts that I use on ForeFlight frequently and I would love to use them on Enroute as well.

The file format of basemaps could be anything that suites the app structure the most (which might already be mbtiles).

An option to select a chart (or even multiple custom charts at the same time) would be required in this case.

I can provide some charts for testing if neccessary.

As an idea it could also be interesting to be able to display kml or geoJson files in case you have special information you want to display on the map.

Love the app so far :)

kebekus, 2021-06-25:
Dear manuuuu400,

thank you for your suggestion! I believe that this feature should be implemented. However, since I program the app alone and in my spare time, it will probably be fall before I can work on an implementation in earnest. To begin, I would appreciate it if you could perhaps send me a few of the MBTILES that you have in mind. For me, it would be easiest if these tiles covered Germany (but this is only "nice to have", not mandatory).

Best,

Stefan

manuuuu400, 2021-06-25:
Sure, I'll send you some. Thanks for considering my suggestion.

Happy flying.

kebekus, 2022-05-22:
@manuuuu400 Thank you for your report. The feature that you requested (=custom base maps) is implemented. I have uploaded v2.16.90 to the Google Play Beta Channel. If you are a beta tester, please have a look. I appreciate your feedback!

closed 2022-05-22 by kebekus