Akaflieg-Freiburg/enroute #188: Hangs on start while connected to WiFi

erik-mueller, 2021-06-27:
**Describe the bug**
The app hangs while starting when connected to WiFi on my Oneplus Nord device. When switching WiFi off, the app starts normally. After the app has started correctly, WiFi can even be switched on again. So it seems to be related to the startup.

**To Reproduce**
Open the app on a Oneplus Nord device while connected to a WiFi network.

**Expected behavior**
The app correctly starts and asks to install maps.

**Screenshots**
Screenshot of the Android bug report attached. Maybe that helps.
[image]

**Smartphone**
 - Device: Oneplus Nord (Model AC2003)
 - OS: Android 11 (Build AC2003_14_210519)
 - Versions 2.8 & 2.8.1 tested

kebekus, 2021-06-27:
Dear Erik,

thank you for your report! I am working on a solution. Unfortunately, none of my devices shows the behavior that you describe. When I come up with a solution, would you be willing to install a test version of the app? I would create an APK-file and send it to you.

Your help is greatly appreciated.

Best,

Stefan.

erik-mueller, 2021-06-27:
Sure, I would like to test out an APK for you.

kebekus, 2021-06-27:
Thanks! I will get back to you as soon as I have something.

kebekus, 2021-06-28:
Dear Erik Müller,

as I wrote, I have tried to fix the issue and produced an APK file, which is available for download here:

- https://nextcloud.cplx.vm.uni-freiburg.de/index.php/s/L3DAfadajz4wB8z

Fixing a bug based on guesswork is of course always difficult. I would greatly appreciate it if you could try the APK and check if it solves the issue, or if it t least changes the behavior of the app on startup.

Best

Stefan.

erik-mueller, 2021-06-28:
Works! I really appreciate your fast support. I guess you will merge this fix then in the main release? Just out of curiosity: What was the cause?

kebekus, 2021-06-28:
Dear Eric,

Thank you! This is very good news.

I just uploaded a public beta (=2.8.2) to Google Play, it should become available in a couple of hours. Would it be possible that you install that version and try it out? I would then make it available to the world once I get your feedback.

The cause: the app needs to monitor the WiFi status, in order to connect to traffic data receivers as soon as the device connects to the traffic data receiver's WiFi. On some devices, the OS seems to report WiFi status before the program has started and before the data is initialized...

Best

Stefan.

erik-mueller, 2021-06-28:
Hi Stefan, I installed the upstream beta version 2.8.2 and this version also works on my device. Thank you for explaining the cause, this explains why it was still working on a different device.

closed 2021-06-29 by kebekus