Akaflieg-Freiburg/enroute #191: Waypoint Edit Disfunctional

woodwind-enr, 2021-06-29:
User feedback:

seit einiger Zeit lassen sich die waypoints im Routing nicht mehr umbenennen. Der Stift erscheint, das Label lässt Sicht editieren wird aber nicht gespeichert (bleibt bei "Waypoints")

Eigene Feststellung:

Beim Nachstellen des Fehlers in der V. 2.8.2 verschwand nach dem Edieren des Wegpunkts der Stift ganz; eine weitere Umbenennung war nicht möglich.

kebekus, 2021-06-30:
Will investigate. There were many internal changes in v2.8.0, which caused much more trouble than I had imagined...

kebekus, 2021-06-30:
I can reproduce the problem. A solution is underway.

kebekus, 2021-07-02:
Thanks for the report. The problem is fixed in version 2.8.3, which is available on Google Play now. Please re-open this issue if the problem persists. Thanks again!