Akaflieg-Freiburg/enroute #195: Show NOTAMS for flight route

kebekus, 2021-07-21:
The title says it all…

kebekus, 2023-01-01:
@Ivan081 I am wondering where the app can download NOTAMS. Are you aware of a source that offers automated access to NOTAMS for free? -- This is a non-commerical app, so we cannot pay for the service.

anthraxn8b, 2023-01-02:
VFRnotam is for free, but I do not know if they provide a free API. But asking them is for free ;-)
I guess that they will not provide a free API because they also offer a VFRnavigation app.

An alternative would be to add the possibility to use a payed account of https://applications.icao.int/dataservices/default.aspx (field to enter your personal apikey in enroute...

kebekus, 2023-01-02:
@anthraxn8b Interesting, I was not aware of VFRnotam. Would you be willing to contact people there and ask? As I wrote earlier, I am coming to my limits with the continued development and the running of the infrastructure…

Best wishes · Happy New Year,

Stefan.

anthraxn8b, 2023-01-02:
I will do my very best ;-)

charlylima, 2023-01-07:
Hello Pilots, 

I am also interested in showing NOTAMS in the app. I would prefer if the notams would be shown on the map. Therefor I already researched that topic a bit some time ago:

The FAA has a NOTAMS API which can create a list of notams in one of 3 possible formats: 
* aixm
* aidap
* geojson

And i think geojson should be possible to show on the map, isn't it?
The API is free to use, but you will need to register for an API key in order to access the data. Not sure if we can use that in an open source software? Or maybe kebekus would need to add the API key every time before compiling and releasing the software?

https://api.faa.gov/s/communityapi/a083t00000V8zfBAAR/faa-afgnotamsapi

You send a request like this:
`https://external-api.faa.gov/notamapi/v1/notams?responseFormat=geoJson&icaoLocation=LOWS`
You also need to have the API key as "client_id" and "client_secret" in the HTTP header. 
The reply is like:

```
{
    "pageSize": 50,
    "pageNum": 1,
    "totalCount": 1,
    "totalPages": 1,
    "items": [
        {
            "type": "Feature",
            "properties": {
                "coreNOTAMData": {
                    "notamEvent": {
                        "scenario": "6000"
                    },
                    "notam": {
                        "id": "NOTAM_1_66426337",
                        "series": "A",
                        "number": "A2810/22",
                        "type": "N",
                        "issued": "2022-12-11T15:20:00.000Z",
                        "affectedFIR": "LOVV",
                        "selectionCode": "QFAXX",
                        "traffic": "IV",
                        "purpose": "BO",
                        "scope": "A",
                        "minimumFL": "000",
                        "maximumFL": "999",
                        "location": "LOWS",
                        "effectiveStart": "2022-12-11T15:20:00.000Z",
                        "effectiveEnd": "2023-03-11T22:00:00.000Z",
                        "text": "SEASONAL CHEMICAL DEICING OF RWY, TWY, APN ADDITIONALLY WITH KFOR\n(FLUID) A",
                        "classification": "INTL",
                        "accountId": "LOWWYNYX",
                        "lastUpdated": "2022-12-11T15:20:00.000Z",
                        "icaoLocation": "LOWS",
                        "coordinates": "4748N01300E",
                        "radius": "005"
                    },
                    "notamTranslation": [
                        {
                            "type": "ICAO",
                            "formattedText": "A2810/22 NOTAMN\nQ) LOVV/QFAXX/IV/BO /A /000/999/4748N01300E005\nA) LOWS B) 2212111520 C) 2303112200 EST\nE) SEASONAL CHEMICAL DEICING OF RWY, TWY, APN ADDITIONALLY WITH KFOR\n(FLUID) A"
                        }
                    ]
                }
            },
            "geometry": {
                "type": "GeometryCollection",
                "geometries": [
                    {
                        "type": "Point",
                        "coordinates": [
                            13.0,
                            47.8
                        ]
                    }
                ]
            }
        }
    ]
}
```

Another Example, 254 Notams around Munich: 
`https://external-api.faa.gov/notamapi/v1/notams?responseFormat=geoJson&locationLongitude=11.57&locationLatitude=48.13&locationRadius=100&pageSize=1000`

[notams-response.txt](https://github.com/Akaflieg-Freiburg/enroute/files/10367152/notams-response.txt)

kebekus, 2023-01-08:
@charlylima I was not aware of this API, thank you so much! I will definitively look into this as soon as possible. I am however, presently 100% preoccupied with porting the app to a new version of the underlying library. Realistically, I would be expect to start working on NOTAMS in early February…

anthraxn8b, 2023-01-08:
@kebekus while I was on holiday and CL has a free and fitting API, I guess it makes no sense to ask the VFRnotam people - because as said, they also provide an application like enroute and so would torpedo their own APP.

kebekus, 2023-01-08:
@anthraxn8b Agreed.

kebekus, 2023-02-07:
This might also be an option: 

- https://www.autorouter.aero/wiki/api/notams/

kebekus, 2023-02-15:
@charlylima I have applied to the FAA for an API key, but it seems it takes forever to process my request, and development is stalled. Do you have a key I could use for development and testing until the FAA is ready? My e-mail address is in the app under "Main Menu/Information/About Enroute…/Authors".

@charlylima If you have concerns, do not hesitate to say "no." I won't be insulted.

kebekus, 2023-03-02:
Implemented. I hope to release a final version within a week or two.

@charlylima @anthraxn8b Thanks for all your help!

closed 2023-03-02 by kebekus