Akaflieg-Freiburg/enroute #196: Flight routs issue

jklopot, 2021-07-23:
Hello,
Briefy -  reinstallation of Enroute resulted with no access to formely defined routes whereas they exist in old folder.  An import of routes from the Library level seems to be disable from the popup menu. Yes, there is a path defined and visible after clicking an information option from menu but I have honestly no idea how to access/enter this path while using file manager to be able to move existing routs from the old directory (no address field visible in file manager).
May I have therefore your assistance please?

Tkank you in anticipation.
Kind Regards,
Jarek

kebekus, 2021-07-24:
Dear Jarek,

you cannot access the flight paths with a file manager. With the new "scoped storage" policies of newer Android version, I do not seem to be able to store this kind of data in publicly accessible directories. I have experimented quite a bit, but I have not found a solution that will work on all devices.

I am closing this issue because I am not aware of any workaround. If you know Android well and have a solution, then please let me know.

Best,

Stefan.

closed 2021-07-24 by kebekus

jklopot, 2021-07-26:
Thank you very much Stefan!

However, it doesn’t explain why the import of flight routes (previous backup files created within Enroute prior to system changes) is not possible – the import option from popup menu is dimmed and disabled in “Library” level – only “Information” works.

 

Regards,

Jarek

kebekus, 2021-07-27:
Dear Jarek,

I think I understand your problem now.

- To import a flight route on Android, transfer the file to Android (e.g. by sending yourself an e-mail, or via Dropbox, …) and then open the file (e.g. in your e-mail program, in a file manager, …). The system will then ask what program should be used to open the file, and "Enroute Flight Navigation" should be one of the choices.
- The "Import" option from the popup menu in the flight route library is used only in the Desktop version of this app. I understand that this might cause confusion, and so I will remove the menu item in the next Android version.

Can you please confirm that this solves your issues now?

Best,

Stefan.

jklopot, 2021-07-27:
Dear Stefan,

 

Big thank You. This recent answer directly corresponds to my former question/concern. However I don’t need to send my flight route files to android because they all have been created by your Enroute application in Android environment before – and they are accessible through file manager in visible directory there. Hence my surprise that now each new route which is created in Enroute application is not located in the same directory anymore.

Anyway good job!

 

Kind Regards,

Jarek

jklopot, 2021-07-28:
Dear Stefan,

 

I promise not to disturb you anymore. 

But, I missed one question while writing to you last time.

With reference to what you described I’d like to ask about Desktop version – how to get this Desktop application please.

 

Thank you in anticipation.

 

Kind Regards,

Jarek

jklopot, 2021-07-28:
Dear Stefan,

 

I quote your instruction:  „To import a flight route on Android, transfer the file to Android (e.g. by sending yourself an e-mail, or via Dropbox, …) and then open the file (e.g. in your e-mail program, in a file manager, …). The system will then ask what program should be used to open the file, and "Enroute Flight Navigation" should be one of the choices.”
Unfortunately the above instruction worked only once – for the first route I’ve been trying to import into my Enroute route library. In case of all the remaining once your Enroute application opened itself after choosing it for opening the *.geojson file – but each time with no imported route on the map (no new route activated on the map).

Kind Regards,
Jarek