Akaflieg-Freiburg/enroute #199: Dist, course and EET to waypoint

meta-sokol, 2021-08-03:
**Is your feature request related to a problem? Please describe.**
Well, not actually a problem, more a very basic function missing to actually use it for navigation 

**Describe the solution you'd like**
Please add the following fields:
- distance to next waypoint
- true course to next waypoint
- estimated enroute time to next waypoint
all calculatedfrom present GPS position 

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

kebekus, 2021-08-06:
Dear meta-sokol,

thank you for your suggestion! The feature is on my radar and will be implemented. However, I am programming the app alone and in my spare time. Development is slow. Unless other people join in to help with the programming (or finance developer), I expect that it will take a very long time before the feature appears.

Best,

Stefan.

kebekus, 2022-03-05:
Dear meta-sokol,

thank you for your suggestion! I am starting the implementation now. Hopefully, I will have a beta version within a few weeks.

Best,

Stefan.

kebekus, 2022-03-15:
Dear all,

The next release of **Enroute Flight Navigation** is coming up, and much has changed internally. I expect a longer-than-usual beta test phase. I would greatly appreciate your feedback if you could try the new features. I am particularly interested to learn if the app reliably recognizes the next waypoint when you follow a route (critical cases might include: routes with sharp angles, return routes, switching the app on and off during the flight, switching the app off before a waypoint, )

Best wishes,

Stefan.

-----

## New features 

### ETE, ETA, and rubber band 

If you fly along a route, the app will show the remaining distance, ETE, and ETA for the next waypoint and the final waypoint. The moving map shows a red "rubber bad" between the current position and the next waypoint. The information is shown under the following conditions:

- The user has set a route.
- The user has set aircraft cruise speed and wind information.
- The aircraft is flying.
- The plane is closer than three nautical miles to the route.
- The distance to the final waypoint is more than three nautical miles.

### Behavior on direct-to

When double-tap into the moving map, the waypoint description dialog will open, and you have the choice to set a quick "direct to" route. The app will now show the rubber band to the waypoint when you do that. The following conditions must be met:

- The user has set aircraft cruise speed and wind information.
- The aircraft is flying.
- The distance to the final waypoint is more than three nautical miles.

### Map update notifications

The app now uses Android notifications to alert you when map updates are available. The alert will not show up while in flight.

kebekus, 2022-03-28:
Dear all,

thank you for your feedback! I understand from all the reports that many users are confused because the Remaining Route Infobox switches on and off during the flight, without apparent reason. Some users also requested that the Infobox is visible on the ground, before take-off.

I have improved the app accordingly and uploaded a new version (=2.15.94) to Google Play. Instead of switching the Infobox off, the app now shows a reason why the box is no longer visible. Also, the box is shown before take-off (except start point = end point, because then the app cannot tell if we are "before start" or "after landing"). If you have the chance, please test and send me your feedback!

Best wishes · Thank you for the help,

Stefan.

kebekus, 2022-04-06:
I am closing this issue because the request is implemented in 2.16.0. Thank you for your help!