Akaflieg-Freiburg/enroute #203: Safesky APP

spuckebaer, 2021-08-20:
Safesky is an app that makes traffic information visible via a mobile network. (OGN, ADS-B & Flarm) An inclusion in the map of Enroute would be fantastic

kebekus, 2021-08-22:
Thank you for the suggestion. I had a look and I am unsure. In principle, this looks like a good idea. I am unsure about two points though.

- I wonder if the network service is fast enough. If traffic information comes late, then it might be worse than having no information at all.
- The service is free for now, but that might change. The setup is "ready to be commercialized". To access it, one needs an API key. This key cannot be included in the source code, so the software is not 100% free anymore. The company behind Safesky can make the service commercial anytime. 
 
All in all, it looks as if the Safesky company starts an attempt to commercialize OGN data. I wonder if perhaps it might be a better idea to use OGN data directly?

Best,

Stefan.

sugacube, 2022-07-14:
> The service is free for now, but that might change. 

I had an introduction even before they went 'live'. iirc, their intention was to provide a "dual model" - free traffic date, but need an subscription for added services (e.g. track friends, messaging, . . .). Still limited as no mode-s traffic shown.

> If traffic information comes late,
I tried it a couple of times. If you have cell coverage it's great. if no coverage, no traffic information of course

FYI, i just asked question on their forum if they plan integration with enroute app.
(just discovered this app today - it looks great. I wasn't aware of its existance - definitely needs more marketing ;-).

kebekus, 2022-08-27:
Dear all,

after some deliberation, I decided against supporting SafeSky for the time being. Apart from all concerns about the business model, some experiments in GA aircraft showed that at least in Germany mobile internet reception is not nearly reliable enough. I understand that this might not be what you wanted to hear. I regret that I have no better news and ask for your understanding.

Best,

Stefan.

closed 2022-08-27 by kebekus